diyruz/freepad #14: PM3 and no PM3 the difference

qlwz, 2021-01-03:
NO PM3 is PM1 OR PM2？

PM3 is PM2 OR PM3？

nurikk, 2021-01-04:
No PM is PM2

qlwz, 2021-01-04:
PM3 need rejoin ?
the first key press is slow

==============================log=======================

debug 2021-01-04 01:18:00: Device '0x00124b000adb81f7' announced itself
info  2021-01-04 01:18:00: MQTT publish: topic 'z2m_2_b/bridge/event', payload '{"data":{"friendly_name":"0x00124b000adb81f7","ieee_address":"0x00124b000adb81f7"},"type":"device_announce"}'
info  2021-01-04 01:18:00: MQTT publish: topic 'z2m_2_b/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b000adb81f7"},"type":"device_announced"}'
debug 2021-01-04 01:18:02: Received Zigbee message from '0x00124b000adb81f7', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 7 with groupID 0
info  2021-01-04 01:18:02: MQTT publish: topic 'z2m_2_b/0x00124b000adb81f7', payload '{"action":"button_7_single","battery":15,"elapsed":1014428,"linkquality":0,"voltage":2700}'

sharathbk1912, 2022-03-21:
Hi @qlwz , with pm3 i am seeing after a while it repairs and it misses the keypad event as such 

[image]

were you able to resolve this ? also which coordinator firmware are you using ?