diyruz/freepad #15: how to compile

qlwz, 2021-01-05:
Z-Stack Home 1.2.2a.44539 no find bdb*.h

nurikk, 2021-01-05:
Use z-stack 3.0.1, don't forget to apply .patch from root folder of this project

bezonar, 2021-01-13:
When trying to compile getting an errors
[Su007]: Unable to open the command file "D:\Zigbee\freepad\CC2530DB\..\..\..\Tools\CC2530DB\f8wEndev.cfg" C:\Users\user\AppData\Local\Temp\EWE91B.tmp 1

bezonar, 2021-01-13:
It would be great to add a note to readme that project should be placed in folder Z-Stack 3.0.2\Projects\zstack\HomeAutomation\

nurikk, 2021-01-13:
> It would be great to add a note to readme that project should be placed in folder Z-Stack 3.0.2\Projects\zstack\HomeAutomation\

feel free to raise PR with this changes

nurikk, 2021-03-07:
https://zigdevwiki.github.io/Begin/IAR_install/

closed 2021-03-07 by nurikk