diyruz/freepad #17: z2m and LeTV

AlexeyHimki, 2021-02-01:
DIYRuZ_FreePad_LeTV.hex Release 2.0.7
zigbee2mqtt-1.17.0

The problem:

Zigbee2MQTT:error 2021-02-01 11:04:26: Failed to interview '0x00124b000adb260f', device has not successfully been paired

AlexeyHimki, 2021-02-01:
After update to 1.17.1:

Zigbee2MQTT:error 2021-02-01 23:31:55: Failed to configure '0x00124b000adb260f', attempt 1 (Error: Bind 0x00124b000adb260f/1 genPowerCfg from '0x00124b0018e2720a/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))

Ich-Eben, 2021-02-05:
I had a similar issue. Desoldering the LED worked for me. It seems that the LED draws to much current sometimes.

spasvisual, 2021-04-15:
> 
> 
> DIYRuZ_FreePad_LeTV.hex Release 2.0.7
> zigbee2mqtt-1.17.0
> 
> The problem:
> 
> Zigbee2MQTT:error 2021-02-01 11:04:26: Failed to interview '0x00124b000adb260f', device has not successfully been paired

Did you fix the problem?