diyruz/freepad #22: It's not a bug, it's a feature

1N4148, 2021-03-25:
I've made a small PCB that fits under a legacy wall switch into the wall box. If it's interesting I'll share the gerber files for the PCB

[image]

nurikk, 2021-03-26:
Hi! PR are always welcome! You can upload files into hardware directory 

https://github.com/diyruz/freepad/tree/master/hardware

jmvaz, 2021-09-30:
Nice!!! Can you share the PCB? Whats the purpose of the switch on board? Is the pairing switch?? It would be nice to be able to pair the device without disassemble the wall plate!

1N4148, 2021-09-30:
These are my Eagle 7.5.0 Files. Have fun :)

[E18-MS1.zip](https://github.com/diyruz/freepad/files/7261823/E18-MS1.zip)