diyruz/freepad #23: how to flash?

boelle, 2021-08-08:
is https://zigdevwiki.github.io/Begin/IAR_install/ stille the guide on how to compile?

i want to make 5 button and 8 button remotes

how do i change settings before i compile the firmware? any english guides?

nurikk, 2021-08-09:
You don’t need a custom firmware for your case, just maintain current matrix schematic and do nothing else

boelle, 2021-08-09:
oki, 

does it only work with zigbee2mqtt? i use deconz/phoscon

nurikk, 2021-08-09:
Afaik it’s supported by dezonz

boelle, 2021-08-10:
i put this in: https://github.com/blakadder/zigbee/pull/371

just so it can get an "official" approval mark

have you tested this yourself?

nurikk, 2021-08-10:
Oh no, I've mistaken freepad repo with flower repo. Freepad isn't supported buy deconz

boelle, 2021-08-10:
that is a shame, would have been perfect for my project

an 8 button version of this

[image]

AnthonMS, 2021-10-01:
A damn shame it's not supported in DeConz! As a fellow Dane, I would have loved an open source project like that!