diyruz/freepad #28: Battery drain - LeTV

rafal83, 2021-12-12:
Hi,

I've flashed LeTV with "DIYRuZ_FreePad_LeTV_PM3.hex", device is used with direct binding to bulb, 5-10 switchs per day, no more 2 weeks with new battery :/

Any idea ?

rafal83, 2021-12-22:
Every 10-12s, wake up, eat 2mA  and sleep, in sleep mode 0.001mA

Consider, only wakeup, 2mA during 1ms, 
in 1 hour there is 6*60 => 360 wake up => total consumtion => 0.72mA
in 1 day => 0.72 * 24 => 17.28 mA, 

CR2032 has 230mA... 13 days of life

Why there is wake up every 10s ?

dxaudio, 2022-01-06:
I have noticed same behaviour. Cant use it as remote at all. Is it possible to change wake up time from 10s to 1h for instance?

mortenmoulder, 2022-01-11:
I can confirm the same thing, however, it seems like every 15 seconds it wakes up, then 10 seconds later it also wakes up, then wakes up 5 seconds later and repeats.

15 seconds wake up: ~300 milliseconds and uses ~2.8 mA
10 seconds wake up: ~40 milliseconds and uses 500 µA

Average power consumption: 150 µA

However, it only does this for 3 minutes, then it goes back to using ~85 µA on average.

Here is what 8 minutes look like including providing power at 00:00:

[image]

Here is what 4 minutes look like **AFTER** 3 minutes:

[image]

Here is what 1 minute and 15 seconds look like in the **FIRST** 3 minutes:

[image]

mortenmoulder, 2022-01-17:
I grabbed a brand new chip and even with a MCP1700 LDO, I got it down to ~4.7 µA after the first 3 minutes. No clue why a new chip worked out - maybe the old was faulty?

It still wakes up every 10-15 seconds for the first 3 minutes, but an average power consumption of 4.7 µA is not bad at all.

[image]

cryoutsolutions, 2022-01-26:
same problem, after two weeks even if I don't use the remote at all, the battery is drained. I use the pm3 version.

maarten682, 2022-09-01:
Any news on this? 2 weeks battery life kinda sucks....

cxw0102, 2022-11-10:
used pm3  ,

> 我拿起了一个全新的芯片，即使使用了 MCP1700 LDO，在前 3 分钟后我还是把它降到了 ~4.7 µA。不知道为什么一个新芯片能成功——也许旧芯片有问题？
> 
> 前 3 分钟它仍然每 10-15 秒唤醒一次，但 4.7 µA 的平均功耗一点也不差。
> 
> [image]
With ubiqua packet capture, it can be seen that the device will always send Data Request without entering PM3 mode.

countrysideboy, 2022-11-10:
> used pm3 ,
> 
> > 我拿起了一个全新的芯片，即使使用了 MCP1700 LDO，在前 3 分钟后我还是把它降到了 ~4.7 µA。不知道为什么一个新芯片能成功——也许旧芯片有问题？
> > 前 3 分钟它仍然每 10-15 秒唤醒一次，但 4.7 µA 的平均功耗一点也不差。
> > [image]
> > With ubiqua packet capture, it can be seen that the device will always send Data Request without entering PM3 mode.

The old version of hardware maybe need extra registers operation to turn the chip into deep sleeping mode.

Flavien, 2023-03-09:
@mortenmoulder So you got a new identical E18-MS1-PCB and its average consumption was lower than the first one you bought?

How is the battery life with the chip that draws 4.7 µA average?

mortenmoulder, 2023-03-09:
@Flavien I actually just installed FreePad PM3 on a brand new chip for a new project. No LDO or 18650 battery, but a CR2032 instead.

With power consumption of 4.7 µA average, you're looking at years of power with a 18650 battery.

Flavien, 2023-03-09:
The LDO was necessary because the 18650 battery is 3.7 V, is that right?

Do you think the problem was the problem caused by the use of a 18650 battery with LDO, or just that the first chip was defective? Or maybe there was a problem with the assembly of your first board?

mortenmoulder, 2023-03-09:
@Flavien The LDO was necessary because of the 18650 battery's high voltage, yeah. About 4.2 when fully charged.

I've had good and bad experiences with these EByte chips. I spoke with some very clever people, including the manufacturer, and it seems like the problem could be related to left-over flux residue, that needs to be cleaned on the outside with isopropyl alcohol or similar, but ultrasound (or something) on the inside of the shield. But, a low quality LDO could definitely also be the cause of this. If you heat it up ever so slightly too much, it could make it less efficient.

I've yet to test the actual power consumption of this new assembly, but I do know for a fact, that getting a PCB manufactured and you soldering the parts on, makes the power consumption go down, compared to leaving everything on a breadboard with jumper wires. I've tested that a bunch of times with these chips thus far.

Flavien, 2023-03-10:
@mortenmoulder Ok, that's interesting. The author of [this video](https://www.youtube.com/watch?v=w-uWYaFztoQ) got the PCB manufactured and soldered the parts, and didn't use a LDO, and he still reported in the comments that the best he got was 3 months of battery life...

It would be good to understand what is causing this. It looks like some people are affected and some are not.

alex2beard, 2023-08-30:
Очень быстрый разряд батареи ни удалось победить ? Два года проблеме )) Ни у кого не завалялось схемы выводов СС на кнопки ? Или конфига для PTVO ?

lych, 2023-09-15:
The same problem. I have two Freepads v1. Battery life is about 2-3 weeks on both.

danielducu, 2024-02-22:
2024 , anyone found solution to the battery problem on this ? Not getting more then 1 month of battery.