diyruz/freepad #29: Stability in Zigbee2MQTT

leoujz, 2022-01-24:
Does anyone meet the stability problem in Zigbee2MQTT? I flash the firmware to CC2530(E18-MS1-PA1), and it seems that very unstable. 

(1) I use coordinator cc2652p, and very close to the 20 keys freepad v1, pcb antenna and ipex antenna all tried.
(2) it's hardly to be configured and interviewed, and only success for a few times in dozens of tries. 
(3) after interviewed, press the button and no signal received by zigbee2mqtt
(4) in device exposes page, click multifunction for each key, somtimes success sometimes fail
(5) in devices binding/reporting setting page, sometimes set it successfully, and the button starts to work, report single/double/triple, and etc.
(6) but suddenly no signal received anymore, sometime power off will make it work again, but more times need to reset and pair the device again

Does anyone meet the same situation?  Is it possible that CC2530 is not powerful enough to run z-stack 3.0?

johanvh83, 2022-03-03:
Hi, I just flashed the E18-MS1-PA1 and use the same coordinator CC2652P. The module was about 1 from the coordinator. After a first try it was interviewed in Z2M and appears in the device list with all 20 buttons, link quatlity and battery information.
I'm able to setup each button (switch type and switch action). These 'entities' also appear in Home Assistant but I don't know how to be able to see the state change of a button when it is pressed. Used the MQTT dump feature in developer tools but the output file remains empty when I pressed the button...

leoujz, 2022-03-08:
I use external usb cable as a power supply. I did not use Z2M in HomeAssistant, I use a standalone Zigbee2MQTT version running on an arm64 board. I'll try that later

phoenix1589, 2022-03-19:
I've attempted to make a handful of these, with chips from 2 different orders. In all cases, the keypad is consistently able to pair with zigbee2mqtt (Conbee_2 coordinator).

Some of the buttons (1, 10-16, 20) work and appear in the zigbee2mqtt logs. The other buttons do not appear in the logs, however the LED does flash when I press them.

Changing the actions in exposes always works; however consistently if the keypad is not close to the coordinator I need to press a button on the keypad a moment after making the change in zibgee2mqtt. This consistently works, although seems a little strange to be needed.

If I bind one of the 'non-working' buttons to a different device; then the button does work to control the device - I still can't get it to log in zigbee2mqtt (which is what I want, as I want it all done via home assistant).

>(5) in devices binding/reporting setting page, sometimes set it successfully, and the button starts to work, report single/double/triple, and etc.

What specific setting do you set in binding or reporting to get it to start to log in zigbee2mqtt? 

Do you see different behaviour with different buttons like O do?

leoujz, 2022-03-27:
yes, I do see  different behavior with different buttons, and for the same button, it has different behavior in many tries of paring process. I think sometimes the configuration might not success for all 20 buttons, some buttons failed but did not fail all times(in my dozens of tries, it only successes 1-2 times). 
I debug the source code, I think all the code is running as expected, I think there must be something wrong with ZStack code running on CC2530. I heard from other people that ZStack 3.0 is hard to running on CC2530 because of limited hardware resource. But I'm not professional on Zigbee, just a guess.

phoenix1589, 2022-03-27:
I guess it's possible. I've read through the code and can't see anything that would treat the buttons differently but for me I get consistent behaviour every time across all 4 units I've made.

There is the original firmware on the modkam.ru site which might be worth a try.