diyruz/freepad #30: How to flash E18-MS1-PCB?

johanvh83, 2022-01-24:
Hi, I want to reproduce your Zigbee keypad. Ordered E18-MS1-PCB.
I was able in the past to flash a CC2652RB USB stick with a .hex file.
As far as I understood I should flash the E18-MS1-PCB with the DIYRuZ_FreePad.hex file.
How can I connect the E18-MS1-PCB to my PC? 
Do I need a USB to Serial TTL convertor like the CH340? https://nl.aliexpress.com/item/32761423124.html?gatewayAdapt=glo2nld&spm=a2g0o.9042311.0.0.27424c4dkahs6F
If yes, could you please clarify where to connect the RX and TX line to the E18-MS1-PCB.
If I'm complety wrong, please give me a hint how to put the .hex file in the E18-MS1-PCB.
Thanks in advance!

phoenix1589, 2022-02-27:
I followed https://www.zigbee2mqtt.io/guide/adapters/flashing/alternative_flashing_methods.html, using the 5 'debugger' pads on the board. 

 P2.1 as DD and P2.2 ad DC. Rest of pins as labelled in that page

johanvh83, 2022-03-03:
Thanks, used also some other websites and ordered and used the CC Debugger.

closed 2022-03-03 by johanvh83