diyruz/freepad #32: Only some buttons work (LED always flashes)

phoenix1589, 2022-02-12:
I've built and flashed a keypad with 20 buttons (DIYRuZ_FreePad.hex). This has paired fine with zigbee2mqtt, and pressing every button I always get a LED flash however only some of the buttons actually have an action received in zigbee2mqtt - specifically buttons 1, and 10-16.

As the LED flashes, I don't see how it could be bad soldering.

Anyone seen this before / have any ideas why only some work?

p1ne, 2023-10-30:
@phoenix1589 any successful fixes of that? I've caught my non-actionnable buttons yesterday - 15-16, 19 in my case

phoenix1589, 2023-10-30:
I've not looked at this in ages, and I never got it working properly.

I have a vague recollection of rebinding some buttons to OnOff and then making a slight change to zigbee2mqtt to cope with just on and off - but I didn't work out how to do that change properly / nearly